intel/IA-Hardware-Composer #243: Allocating unneeded large buffers

kalyankondapally, 2018-02-18:
Tracing in overlaylayer shows we are allocating buffers of large size which fail fb creation:

SetBuffer: buffer 2880 2560   -> Buffer size
02-18 20:21:11.651  1969  2149 E iahwcomposer: SetBuffer: layer 1920 1080  -> Frame rect

This might be an alignment issue in buffer manager side.

kalyankondapally, 2018-02-21:
seems to be done when incoming layers have size of 1920, 1080. We should track down in Minigbm side

harishkrupo, 2018-02-21:
What about the behavior with GBM?

kalyankondapally, 2018-02-21:
not sure

harishkrupo, 2018-02-21:
Ok, will check with GBM.

kalyankondapally, 2018-02-21:
Also, probably when nativehandler is initialized, we should set the maximum width and height and limit it to them when allocating memory.

harishkrupo, 2018-02-21:
yeah, that can be done. But even then if the maximum W/H is 1920x1080 we will still end up allocating the same?

kalyankondapally, 2018-02-21:
if requested size is larger than this, we should limit it to this size

harishkrupo, 2018-02-21:
Ran the linux_test app with GBM. it is allocating properly. Need to check if this is a minigbm issue or Android side allocation.

kalyankondapally, 2018-04-25:
seems like Android FW is requesting these.

closed 2018-04-25 by kalyankondapally