intel/IA-Hardware-Composer #258: Fix warnings.

kalyankondapally, 2018-03-07:
We shouldn't have any warnings on Android builds.

kalyankondapally, 2018-03-07:
@ravelar is going to look at this.

stripes416, 2018-03-12:
@kalyankondapally I think you mean @csravelar. I'm not sure who 'ravelar' is?

kalyankondapally, 2018-03-12:
@stripes416 He has two id's :)

csravelar, 2018-03-12:
My intel given github.intel.com id is ravelar. My github is csravelar. Looks like we tagged some random guy hehe.

kalyankondapally, 2018-03-23:
1 warning generated.
[ 31% 14/44] target  C++: libhwcomposer_common <= vendor/intel/external/android_ia/hwcomposer/common/core/gpudevice.cpp
vendor/intel/external/android_ia/hwcomposer/common/core/gpudevice.cpp:581:17: warning: comparison of integers of different signs: 'int' and 'size_t' (aka 'unsigned long') [-Wsign-compare]
      if (index < num_displays) {
          ~~~~~ ^ ~~~~~~~~~~~~
1 warning generated.
[ 38% 17/44] target  C++: libhwcomposer_common <= vendor/intel/external/android_ia/hwcomposer/common/core/nesteddisplay.cpp
vendor/intel/external/android_ia/hwcomposer/common/core/nesteddisplay.cpp:190:54: warning: unused parameter 'source_layers' [-Wunused-parameter]
bool NestedDisplay::Present(std::vector<HwcLayer *> &source_layers,

still seeing some warnings

closed 2018-04-25 by kalyankondapally