intel/IA-Hardware-Composer #268: Clean up erros returned by HWCServiceAPI

kalyankondapally, 2018-03-15:
Currently, we return -1, instead we could define our error values and return them. Suggestion by Kevin.
"Given that we are using status_t return type, we ought to use a real return code. I'd suggest BAD_VALUE."

MicheleL-Intel, 2018-03-15:
I will handle it.

strassek, 2018-03-28:
Thanks for the proposal @MicheleL-Intel !

> 1. New hwcs_status_t replaces ALL "android::" references. Doing this will harmonize all variances in the current implementation, which currently calls out the status codes in various forms. Doing this will enable more consistent references of the status codes.

Do we have any references to `android::` outside of the Android specific code in os/android/? I'm definitely all for consolidation, but in instances where we are providing a public API to the OS we need to be sure to play nice and use the return types and status codes that are expected. So maybe that means translating an internal code to the `android::status_t` equivalent just before returning.

> This new enum also renames HWCS_TRUE to HWCS_ENABLE, and HWCS_FALSE to HWCS_DISABLE since it is a more appropriate name for its purpose and usage (see HWCService_Widi_GetSingleDisplay()). Let me know if you disagree and I'll preserve the same name.

I'm not so sure wrapping these values into the hwcs_status_t enum is such a great idea. In the specific case of HwcService_Widi_GetSingleDisplay(), the return type is used to indicate if something went wrong, as where the 'enable' param actually answers the question "Is the display enabled?". Using a bool type there guarantees that the caller will only have to handle 'true' or 'false' cases. Using separate enums pins down the expectations of the API.

If you still think it's a good idea then go ahead and try it out...

> 2. Combine existing Linux and Android error codes into one since they're exactly the same. A new enum type called hwc_error_t will be created under new "common/include" folder. The Makefile will be modified accordingly.

It is true that today most of the Android error codes are defined in terms of Linux error codes, but I'm not sure if we want to depend on that arrangement.

MicheleL-Intel, 2018-03-28:
Thank you for reviewing @strassek 

1. Do we have any references to android:: outside of the Android specific code in os/android/? I'm definitely all for consolidation, but in instances where we are providing a public API to the OS we need to be sure to play nice and use the return types and status codes that are expected. So maybe that means translating an internal code to the android::status_t equivalent just before returning.

[MicheleL-Intel] There aren’t any “android::” outside of the Android-specific code.  This objective is to eliminate “android::” indirections in the code as part of the cleanup.  Each item in the enum will be equated to well-known defines (such as -EINVAL), which is what “android::<enum define>” is currently set to.  In other words, the well-known values will be preserved.

2. Combine existing Linux and Android error codes into one since they're exactly the same. A new enum type called hwc_error_t will be created under new "common/include" folder. The Makefile will be modified accordingly.

[MicheleL-Intel] I agree there’s a risk involved.  This will be omitted.

closed 2018-04-25 by stripes416

MicheleL-Intel, 2018-05-01:
@strassek , I was just assigned this and it has been closed by you 6 days ago.  Is there anything else left to be done?