intel/IA-Hardware-Composer #283: Enable Vulkan backend

kalyankondapally, 2018-03-20:
@strassek  did most of the work but lot has changed in HWC. We might need to fix any bugs which got introduced into Vulkan backend because of these changes and also make it one of the run time options going forward.

Assigning it to @strassek

strassek, 2018-03-21:
I started to update the Vulkan backend and got it building again. Unfortunately I'm hitting a -EBADF error in drm_gem_prime_fd_to_handle(). It's hard to say where things are going wrong, as the GLES backend takes this same path.

strassek, 2018-03-28:
I ran another test with IAHWC and Mesa sources rolled back to where they were when the Vulkan backend originally landed, and I saw the exact same error. I'll have to try again with a more stable kernel.

strassek, 2018-04-03:
It turns out one of the import issues I was having was cased by a regression in Mesa. I have submitted a patch for that:
https://patchwork.freedesktop.org/patch/214687/

closed 2018-09-13 by stripes416