intel/IA-Hardware-Composer #289: HWC Service fails to start.

kalyankondapally, 2018-03-21:
One can see the following errors in logcat:

03-21 12:40:04.743  2109  2109 E SELinux : avc:  denied  { add } for service=hwc.info pid=6084 uid=1000 scontext=u:r:hal_graphics_composer_default:s0 tcontext=u:object_r:default_android_vndservice:s0 tclass=service_manager permissive=0
03-21 12:40:04.743  2109  2109 E VendorServiceManager: add_service('hwc.info',1) uid=1000 - PERMISSION DENIED
03-21 12:40:04.743  6084  6084 E         : Failed to start hwc.info service

stripes416, 2018-03-21:
Hey @kalyankondapally  -- can you post the steps to reproduce this? I'm not sure how you got to this point.

kalyankondapally, 2018-03-21:
@stripes416 : 
1) Sync MR1 code base:
repo init -u git@github.com:android-ia/manifest.git -b omr1_master

2) Build, install and flash (Grub) on APL NUC.
3)Boot to home screen.
4)Connect to network via wifi (I couldn't get Ethernet working).
5)adb shell
6)logcat | grep "hwc" and you should see these messages

pyn1, 2018-03-22:
@kalyankondapally , I had pushed a patch to remove the warning. Unfortunately, it was not merged. I will be pushing the patch for same again today.

kalyankondapally, 2018-03-22:
@pyn1 oh k.

kalyankondapally, 2018-03-22:
I think you are referring to this one right @pyn1 https://github.com/android-ia/device-androidia/pull/116/commits/787322be938f70d8a4cb33c0dfaa6153411ec7ea

pyn1, 2018-03-22:
Yes, you are right.