intel/IA-Hardware-Composer #332: Runtime HSBC adjust is not working

xuguangxin, 2018-04-16:
this is required by PREQ-28918, we need adjust Hue, Sharepness... when we pause the video
please consider add this

strassek, 2018-04-17:
Is this a bug or a feature request? Please provide steps for reproduction.

xuguangxin, 2018-04-17:
From design, we hope we can change hsbc at the pause. Currently, it does not work. You can say it's a bug

step to reproduce:
1. play a  video on Android .
2. pause the video 
3. run 'hwcservice_test -b 100' to adjust Brightness to 100. 
4. you can see no brightness improvement on the screen.
5. unpause the video.
6. now you see the video turn bright

xuguangxin, 2018-04-17:
@kalyankondapally

stripes416, 2018-07-12:
@xuguangxin do you remember what video player you were running the video with? Just native or a specific app?

js0701, 2018-09-10:
https://github.com/intel/IA-Hardware-Composer/commit/21c2f60dfd5044cfa0e640b09d9c42e52a669f00
it may already been fixed by this patch

chunge123, 2018-09-10:
Can't duplicate this issue on  https://buildbot.sh.intel.com/absp/builders/f_p_opensourcegfx_master-engineering/builds/62 build,
step:
1.play a video on Android .
2.pause the video
3.run 'hwcservice_test -b 100' to adjust Brightness to 100.
4.the video turn bright on the screen.

xuguangxin, 2018-09-10:
@chunge123 , could you merge OAM-65128 if this landing on GP2.0?

thanks

jennycao, 2018-09-10:
@xuguangxin @js0701  This patch would be merge to 1A ww37.2.

closed 2018-09-11 by js0701