intel/IA-Hardware-Composer #353: Move shader generation and upload to init path

strassek, 2018-05-03:
Currently we are generating shaders on-demand in the compositor draw path. This means we can't provide consistent draw latency, as compositions may be subject to blocking on shader gen/upload.

Pick a sensible max number of supported layers, and gen/upload shaders in bulk at program init. We will then reject requests for compositions that exceed that number.

Requirements:
 * The interface we use to tell the backend to do the gen/upload should be graphics API agnostic. i.e. for Vulkan we would generate a pipeline, not just the shader modules.

 * The process of loading shader binaries from storage and then uploading to gpu should happen sequentially, such that it becomes very unlikely that the shader binary memory goes cold before upload.

 * Get rid of LOAD_PREBUILT_SHADER_FILE and compile in prebuilt shader support by default

js0701, 2018-11-19:
See glprogram.cpp
the vertex shader and program shader are generated runtime based on the layer_count.
We can't implement this feature if the layer_count is dynamically changed.
We can decide a maximum layer_count, and then use a uniform for the real layercount,
in this case it still possible?

js0701, 2018-11-19:
After investitation, this feature is hard to be implemented by gles3.3 spec.
However it is feasible with gles4.2 afterwards. We can wait for this until mesa implement it