intel/IA-Hardware-Composer #689: Build error with corresponding android10-release minigbm branch

maurossi, 2020-04-16:
Hi,
just a question, when trying to build android10-release branch of this project,
I am using intel/minigbm project same name branch android10-release,
but I am getting the following building error:

```
external/IA-Hardware-Composer/platformminigbm.cpp:20:10: fatal error: 'i915_private_types.h' file not found
#include "i915_private_types.h"
         ^~~~~~~~~~~~~~~~~~~~~~
1 error generated.
```

How do you build android10-release branch of this project?
Mauro

Shao-Feng, 2020-04-17:
Hi,

Could you sync the codes to the latest for Android 10?
this head file "i915_private_types.h" is not used any more.

maurossi, 2020-04-17:
Hi,
thank for response, I synced this project to android10-release branch and its HEAD commit 3eca0df4b63dd5def28a73d8ec123087f7262f43 introduces an include of i915_private_types.h

Which latest for Android 10 branch are you referring to?

Thanks

Shao-Feng, 2020-04-17:
add the one into your repo https://github.com/intel/IA-Hardware-Composer.git
and the last commit is https://github.com/intel/IA-Hardware-Composer/commit/f18ffcd7feba1dee61fa8cd9dffd66bde0c2008b

Use the master branch, don't use the branch android10-release, that one is not released one, it is named incorrectly

maurossi, 2020-04-17:
Thanks a lot

closed 2020-04-17 by maurossi