intel/IA-Hardware-Composer #690: Building errors with BOARD_USES_VULKAN := true

maurossi, 2020-04-19:
Hi,
on current master branch, I'm getting the following building errors with Android Q,
the first and second errors, if they can be considered as false positive on x86 builds, 
may be avoided using -Wno-cast-align, but that does not fix the third build error.

```
external/IA-Hardware-Composer/common/compositor/vk/vkprogram.cpp:88:25: error: cast from 'const uint8_t *' (aka 'const unsigned char *') to 'const uint32_t *' (aka 'const unsigned int *') increases required alignment from 1 to 4 [-Werror,-Wcast-align]
  module_create.pCode = (const uint32_t *)vkcomp_vert_spv;
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
external/IA-Hardware-Composer/common/compositor/vk/vkprogram.cpp:97:25: error: cast from 'const uint8_t *' (aka 'const unsigned char *') to 'const uint32_t *' (aka 'const unsigned int *') increases required alignment from 1 to 4 [-Werror,-Wcast-align]
  module_create.pCode = (const uint32_t *)vkcomp_frag_spv;
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from external/IA-Hardware-Composer/common/compositor/vk/vkprogram.cpp:15:
In file included from external/IA-Hardware-Composer/common/compositor/vk/vkprogram.h:20:
external/IA-Hardware-Composer/common/compositor/vk/vkshim.h:75:14: error: cast from 'uint8_t *' (aka 'unsigned char *') to 'float *' increases required alignment from 1 to 4 [-Werror,-Wcast-align]
      return (T *)ptr_;
             ^~~~~~~~~
external/IA-Hardware-Composer/common/compositor/vk/vkprogram.cpp:225:34: note: in instantiation of function template specialization 'hwcomposer::RingBuffer::Allocation::get<float>' requested here
  float *vert_ub = vert_ub_alloc.get<float>();
                                 ^
3 errors generated.
```

maurossi, 2021-02-14:
Hello @tpalli 

when enabling BOARD_USES_VULKAN with current master branch at commit ac056aa 
I'm still getting the following building error:

```
target  C++: libhwcomposer_common <= external/IA-Hardware-Composer/common/compositor/vk/vkprogram.cpp
FAILED: out/target/product/x86_64/obj/STATIC_LIBRARIES/libhwcomposer_common_intermediates/compositor/vk/vkprogram.o
In file included from external/IA-Hardware-Composer/common/compositor/vk/vkprogram.cpp:15:
In file included from external/IA-Hardware-Composer/common/compositor/vk/vkprogram.h:20:
external/IA-Hardware-Composer/common/compositor/vk/vkshim.h:75:14: error: cast from 'uint8_t *' (aka 'unsigned char *') to 'float *' increases requi
red alignment from 1 to 4 [-Werror,-Wcast-align]
      return (T *)ptr_;
             ^~~~~~~~~
external/IA-Hardware-Composer/common/compositor/vk/vkprogram.cpp:225:34: note: in instantiation of function template specialization 'hwcomposer::Rin
gBuffer::Allocation::get<float>' requested here
  float *vert_ub = vert_ub_alloc.get<float>();
                                 ^
1 error generated.

```

Is there some way to avoid the building error, if not fixing it?

tpalli, 2021-02-14:
IIRC the vulkan rendering backend of IA-Hardware-Composer is 'experimental' and is not used, we do have vulkan support enabled though .. not sure how this is handled in the celadon build.

maurossi, 2021-02-14:
The difference, if I am not wrong, should be that BOARD_USES_VULKAN should enable the Android GUI to use Vulkan API,
while at the moment in android-x86 builds we only install vulkan anv and vulkan radv modules
and Vulkan API is available to Vulkan enabled Apps

Do you know if  github.com/`projectceladon/`IA-Hardware-Composer master branch is built with `BOARD_USES_VULKAN := true ` ?

In case BOARD_USES_VULKAN is unessential for IA-Hardware-Composer, because vulkan compositor backend in IA-Hardware-Composer is optional and can be kept disabled,
I could just rename 'g/BOARD_USES_VULKAN/IAHWC_USES_VULKAN/s'
or it will not work?

maurossi, 2021-02-14:
In any case why is RingBuffer::Allocation::get in common/compositor/vk/vkshim.h incompatible with <float> ?

Mauro

tpalli, 2021-02-15:
> The difference, if I am not wrong, should be that BOARD_USES_VULKAN should enable the Android GUI to use Vulkan API,
> while at the moment in android-x86 builds we only install vulkan anv and vulkan [...]

Ok I see, yeah BOARD_USES_VULKAN is not used by Celadon. You can just rename it, AFAIK it is experimental and unmaintained code.

Shao-Feng, 2021-02-18:
VULKAN is not used and maintained in Celadon.

maurossi, 2021-02-18:
Thanks @Shao-Feng and @tpalli 
Closing the issue

closed 2021-02-18 by maurossi