intel/IA-Hardware-Composer #726: build Celadon will show wrong message

KunYi, 2022-12-16:
build Celadon project with Android 11, got the following wrong message.
```
build/make/core/[binary.mk:953](http://binary.mk:953/): warning: target `dotdot/out/target/product/caas/obj_x86/SHARED_LIBRARIES/libvts_interfacespecification_intermediates/vts-generated/src/lib/ndk/bionic/1.0/libcutilsV1.o' doesn't match the target pattern
[ 88% 244/277] including vendor/intel/external/hwcomposer-intel/Android.mk ...
"PLATFORM_VERSION 11"
"ANDROID_VERSION "
/bin/bash: line 0: test: -ge: unary operator expected
```

because the below only have android 9, missed greate Android 9
https://github.com/intel/IA-Hardware-Composer/blob/efe339ff5aa656e40b43312c133b142ac2cc227f/Android.common.mk#L28-L45

so the below will invoke the wrong message
https://github.com/intel/IA-Hardware-Composer/blob/efe339ff5aa656e40b43312c133b142ac2cc227f/Android.common.mk#L165-L167

Shao-Feng, 2022-12-19:
This component is not used in Celadon R anymore. I will commit a patch to remove it from the SOONG list to avoid compile it

KunYi, 2022-12-19:
okay! sound great!

closed 2022-12-19 by KunYi